rossumur/esp_8_bit #1: Is it possible to hook up multiple wii motes ?

tonym128, 2020-05-22:
I'm thinking of taking your code and seeing about making a takeabout console to friends houses and plugging into their TV's. Do the libraries support mutliple BT Controllers / Wii Motes being hooked up ?

I will take a look, but thought input from you would be a good starting point :)

This is a very awesome project, I pulls so many cool things together. I recently did something similar with just a black and white screen output to TV and a PS3 controller for an Arduboy type console
https://github.com/tonym128/ESP32_Arduboy/tree/tvout

ejri, 2020-05-22:
Not sure if it's possible to connect them simultaneously. You'd have to iterate the connection through them, but typically the fastest scan takes around 1s, and a recommended scan to find all devices would be around 5-10s. Check here,  

https://github.com/nkolban/esp32-snippets/issues/469

rossumur, 2020-05-23:
> nkolban/esp32-snippets#469

You can connect up to 4 wiimotes. Scan on boot is 5 seconds and seems to
always find them on first boot. Subsequent reconnects happen automatically.
As they pair the LEDs reflect if they matched Joystick 1..4
cheers

closed 2020-05-27 by tonym128