rossumur/esp_8_bit #2: What is the resolution & color depth?

bootrino, 2020-05-29:
Hello,

I wrote a comparison of esp32 video solutions here:
https://www.reddit.com/r/esp32/comments/gpezk3/feature_comparison_of_esp32_graphics_sound_game/

I wonder would you mind confirming please, what is the maximum resolution that your solution can do for each of PAL & NTSC?  Also what is the color depth?

As a side question, can you indicate what plans you have from here for this software?

thanks

rossumur, 2020-06-01:
The Atari emulation runs at 336x192. Hardware/Software is capable of color up to 720x576 for PAL, 640x480 for NTSC. DAC resolution x phase x saturation =~ 16k colors.

Running the DAC at the highest rate (20Mhz) in grayscale yields 2048x480 or 2048x576 at 64 levels of gray.

rossumur, 2020-06-01:
Planning on adding emulation for other devices, native gaming apis, videos streaming etc.

bootrino, 2020-06-01:
Just to confirm - are those resolutions theoretical maximums or actual maximums you've got from your software/hardware?  i.e. you got 720x576 PAL and 640x480 NTSC with esp32? Would you mind helping me understand please .... I imagine you'd need more RAM for such resolutions?  

One of the most intriguing ideas that came to mind when I saw your PAL/NTSC solution is you could effectively make the esp32 a graphics controller/coprocessor.  So for example you could have a retro/new machine like an eZ80 CPU that includes an esp32 on the same PCB board as the graphics/wifi coprocessor connected via SPI. Or really any other CPU that needed a TV display controller.

Great work.  Looking forward to seeing where you take it.

rossumur, 2020-06-03:
You don't need a full frame buffer for 720x576, 640x480, 1280x480 etc
Arduinocade did 480p with a total of 2k of RAM in the system:
https://www.youtube.com/watch?v=nGIujZiEu_o
Tiles / Sprites, RLE, sorted lines give you plenty of flexibility without
requiring a frame buffer. Even full 3D works if you use character sets.

You could easily do a nice 160 column display with antialiased monochrome
fonts, for example. The character frame buffer would be <10k.

bootrino, 2020-06-04:
Sounds really awesome.  I love programming like this - it's magic.

Hopefully you'll put up more videos showing some of these things in action.