rossumur/esp_8_bit #4: MALLOC32 FAILED allocation of MEMORY_mem:65540!!!!

bootrino, 2020-06-09:
Hello I'm trying to run this and I get this error:

```
esp_8_bit

mounting spiffs (will take ~15 seconds if formatting for the first time)....
... mounted in 99 ms
frame_time:0 drawn:1 displayed:0 blit_ticks:0->0, isr time:0.00%
emu_task atari800 running on core 0 at 240000000mhz
MALLOC32 245608 free, 113792 biggest, allocating Screen_atari:92160
MALLOC32 allocation of Screen_atari:92160 3FFE4374
MALLOC32 153432 free, 64776 biggest, allocating MEMORY_mem:65540
MALLOC32 FAILED allocation of MEMORY_mem:65540!!!!####################
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698
```
If it helps, this is system information for the esp32 chip:

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5848
entry 0x40080698
Internal Total heap 405012, internal Free Heap 379196
SPIRam Total heap 0, SPIRam Free Heap 0
ChipRevision 1, Cpu Freq 240, SDK Version v3.3-beta1-179-ge931fe9f5-dirty
 Flash Size 4194304, Flash Speed 80000000
ps=0x0
```

I've tried it on 2 different esp32 dev boards but same result each time.

Any suggestions as to what I might be doing wrong?

thanks

bootrino, 2020-06-09:
Sorry I only just saw in the source code that I need to "ESP32 Sketch Data Upload" first.

Still not working yet, but please disregard first error in this thread.  Onto solving the next5 issue.

bootrino, 2020-06-09:
OK so progress made, and some feedback.

It appears that the documentation on the readme page says pin 25 in the schematic but the source code says pin 26.

A suggestion,  if there was a a quick start guide that said: 

1: Check the ping number in src/video_out.h - it must be either pin 25 or pin 26
2: The video cable is the centre pin of the yellow connection on an RCA composite cable - connect this to pin 25 (or pin 26) on the esp32.
3: Connect the outer shield on the yellow cable to one of the ground pins on the esp32

This will get the video displaying.

I managed to get a good clear and stable NTSC signal.  I could not however get a stable PAL signal.

I've put a video here showing - sorry its rather long winded. The upshot of it though is that NTSC worked on both TV's and PAL didn't seem to lock in cleanly - on one TV the signal was rolling and on the other TV the signal was stable but did not look right - with lines and a corrupted border.

Here is the YouTube video:  https://www.youtube.com/watch?v=LtlMDrhE50c

bootrino, 2020-06-09:
Further update - maybe not worth worrying about my PAL issues - I have tried bitluni's composite video on exactly the same hardware and it's janky too - it's not rolling, but it is jumping and tearing a bit.

I'll see if somehow I can get a cleaner composite video plug and cable setup and that might clear up the PAL issues.

RoCorbera, 2020-06-22:
As for PAL, there many PAL modes in the world. I have a PAL-N decoder TV that works fine for me. You may check https://en.wikipedia.org/wiki/PAL for your region/TV decoder. 

I tried ESP_8_bit on a NTSC TV and the image displays exactly as in you TV. It sounds like this software cuts the edges of the image on NTSC, for Chase Game...

One important detail: This is an emulator that can run PAL/NTSC coded cartridge. Some games are olny designed for one of those two systems as they have different pixel resolution. 
Some games may display wrong if the wrong "TV system" is used.

gregsadetsky, 2021-04-03:
Thanks, I found this thread by googling the same error message.

I received this message using the [ESP32 Development Board](https://www.adafruit.com/product/3269) when using the default `#define EMULATOR EMU_ATARI` value. I changed it `#define EMULATOR EMU_NES` and that worked. Something is "wrong"/incompatible by default with the Atari emulator and this board, but at least the NES emulator does boot up.

Second point: you mention pin 25/26, but interestingly enough, that line `#define VIDEO_PIN   26` ([here](https://github.com/rossumur/esp_8_bit/blob/c65fa1e70e3a981abb9f232ef481fcf0a42eab0a/src/video_out.h#L18)) is a red herring -- there is no use of `VIDEO_PIN` anywhere in the code. The video is rather setup on pin 25 as that pin is the DAC output pin. See [here](https://github.com/rossumur/esp_8_bit/blob/c65fa1e70e3a981abb9f232ef481fcf0a42eab0a/src/video_out.h#L137) in the code, and [this pinout](https://cdn-shop.adafruit.com/product-files/3269/pinout_wroom_pinout.png) which shows physical pin 14 being GPIO25 and the DAC_1 pin as well. Oof! :-) Pin 25 for the video signal and any Ground pin worked here on my CRT monitor. Cheers!

OK1FSX, 2022-07-26:
ESP32 library newer than 1.0.4 might have issues when compiling. Downgrade of esp32 library to 1.0.3 fixed my no video problem.

OK1FSX, 2022-07-29:
Please can anybody help? I am trying to make CORNN64 project with microSDcard (setting PAL, EMU_ATARI). Without connecting MicroSDcard adapter I can compile and upload everything successfully, it works. But if I connect MicroSD card module with card and enable it in the code, uploading of code ends with "MALLOC32 FAILED allocation of MEMORY_mem:65540" error and rebooting. Older libraries in Arduino IDE or older ArduinoIDE don´t help. I have read, the problem is concerning esp_heap_caps.h or multi_heap.h libraries and bad allocation of memory,  but I don´t know how to fix it. By the way, I cannot find sources (.c) of these libraries in ArduinoIDE to edit. Where are these sources? Headers (.h) are present but sources not. Thanks.

EmVee381, 2022-12-16:
> Please can anybody help? I am trying to make CORNN64 project with microSDcard (setting PAL, EMU_ATARI). Without connecting MicroSDcard adapter I can compile and upload everything successfully, it [...]

as i am playing with this project also, it is everything most around memory :) if you dont use buetooth then comment out hid* functions in main.cpp, you will save memory and you will pass to next level of this game :)))