rossumur/esp_8_bit #10: Failure when connecting Bluetooth keyboard resulting in Panic/Reset of ESP32

RoCorbera, 2020-06-14:
Hello Rossum,

I really enjoyed this project and I'm impressed by your knowledge and results!
Since your first public project RBox, passing by the Arduinocade and now with the ESP_8_Bit, you demonstrate a passion and impressive skills! Congratulations!

I have deployed the ESP_8_bit to my ESP32 WROOM. It runs each emulator with no issues.
But when I tried to attach a Bluetooth V3.0 keyboard I observer that it starts to display images to the TV but after a few seconds the ESP32 got rebooted.

So I decided to turn on HCI Messages and logged the serial output to help on finding out what is going on. These are the messages I got in one reboting cycle:

========================================================
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

esp_8_bit

mounting spiffs (will take ~15 seconds if formatting for the first time)....
... mounted in 96 ms
> CMD 030C00
emu_task atari800 running on core 0 at 240000000mhz
< EVT 0E0405030C00
HCI_COMMAND_COMPLETE_EVT 4 bytes
 -> HCI_RESET 0C03 No Error
> CMD 051000
< EVT 0E0B05051000FD03FF09000400
HCI_COMMAND_COMPLETE_EVT 11 bytes
 -> HCI_READ_BUFFER_SIZE 1005 No Error
> CMD 091000
< EVT 0E0A05091000DA690BC40A24
HCI_COMMAND_COMPLETE_EVT 10 bytes
 -> HCI_READ_BD_ADDR 1009 No Error
> CMD 240C0307020C
> CMD 1A0C0103
< EVT 0E0405240C00
HCI_COMMAND_COMPLETE_EVT 4 bytes
 -> HCI_WRITE_CLASS_OF_DEVICE 0C24 No Error
< EVT 0E04051A0C00
HCI_COMMAND_COMPLETE_EVT 4 bytes
 -> HCI_WRITE_SCAN_ENABLE 0C1A No Error
> CMD 010405338B9E0500
> CMD 200C0100
< EVT 0F0400050104
HCI_COMMAND_STATUS_EVT 4 bytes
 -> HCI_INQUIRY 0401 No Error
< EVT 0E0405200C00
HCI_COMMAND_COMPLETE_EVT 4 bytes
 -> HCI_WRITE_AUTH_ENABLE 0C20 No Error
MALLOC32 246804 free, 113792 biggest, allocating Screen_atari:92160
MALLOC32 allocation of Screen_atari:92160 3FFE4374
MALLOC32 154628 free, 67336 biggest, allocating MEMORY_mem:65540
MALLOC32 allocation of MEMORY_mem:65540 3FFCF8EC
MALLOC32 89072 free, 48532 biggest, allocating _lines:960
MALLOC32 allocation of _lines:960 40094260
MALLOC32 88096 free, 47556 biggest, allocating under_atarixl_os:16384
MALLOC32 allocation of under_atarixl_os:16384 40094630
MALLOC32 71696 free, 31156 biggest, allocating under_cart809F:8192
MALLOC32 allocation of under_cart809F:8192 40098640
MALLOC32 63488 free, 22948 biggest, allocating under_cartA0BF:8192
MALLOC32 allocation of under_cartA0BF:8192 4009A650
Emu::load failed for /atari800/atari_robot.xex.cfg
video_init
< EVT 020F0141DF3C007320010000402500F13E
HCI_INQUIRY_RESULT_EVT 15 bytes
20:73:00:3C:DF:41:402500 input device added
< EVT 020F0141DF3C007320010000402500F63E
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C0073200100004025000D3F
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C0073200100004025003A3F
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C007320010000402500493F
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C007320010000402500503F
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C007320010000402500663F
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C007320010000402500783F
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C007320010000402500F13F
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C0073200100004025000240
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C0073200100004025001840
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C0073200100004025002E40
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C0073200100004025003E40
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C0073200100004025005F40
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 020F0141DF3C0073200100004025006B40
HCI_INQUIRY_RESULT_EVT 15 bytes
< EVT 010100
HCI_INQUIRY_COMP_EVT 1 bytes
> CMD 05040D41DF3C00732018CC0100000000
< EVT 0F0400050504
HCI_COMMAND_STATUS_EVT 4 bytes
 -> HCI_CREATE_CONNECTION 0405 No Error
< EVT 030B00810041DF3C0073200100
HCI_CONNECTION_COMP_EVT 11 bytes
20:73:00:3C:DF:41 connected on handle 129
> CMD 19040A41DF3C00732000010000
< EVT 0F0400051904
HCI_COMMAND_STATUS_EVT 4 bytes
 -> HCI_REMOTE_NAME_REQUEST 0419 No Error
< EVT 1B0ASSERT_ERR(lm_env.local_name), in lm.c at line 72
3810005
HCI_MAX_SLOTS_CHANGED_EVT 3 bytes
unhandled hci case
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40088278  PS      : 0x00060334  A0      : 0x801258bd  A1      : 0x3ffcd8b0  
A2      : 0x00000001  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffbdbb4  
A6      : 0x00000058  A7      : 0x00000000  A8      : 0x80088278  A9      : 0x3ffcd890  
A10     : 0x00000032  A11     : 0x00000032  A12     : 0x00000010  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffd  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40088221  LEND    : 0x40088228  LCOUNT  : 0x00000000  

Backtrace: 0x40088278:0x3ffcd8b0 0x401258ba:0x3ffcd8d0 0x4002509e:0x3ffcd8f0 0x4011d46d:0x3ffcd930 0x4001a637:0x3ffcd950 0x40019d11:0x3ffcd980 0x40055b4d:0x3ffcd9a0 0x4011a34b:0x3ffcd9c0 0x4011a899:0x3ffcd9e0 0x4008dc49:0x3ffcda10

Core 1 register dump:
PC      : 0x4008eae8  PS      : 0x00060134  A0      : 0x8008f95a  A1      : 0x3ffc9830  
A2      : 0x00000001  A3      : 0x3ffbd868  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060123  A7      : 0x00000000  A8      : 0xb33fffff  A9      : 0x0000abab  
A10     : 0x3ffbab6c  A11     : 0x000001ac  A12     : 0x3ffbab8c  A13     : 0x3ffbe790  
A14     : 0x3fff7574  A15     : 0x3ffe0994  SAR     : 0x00000010  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

Backtrace: 0x4008eae8:0x3ffc9830 0x4008f957:0x3ffc9850 0x4008d31a:0x3ffc9870 0x4008d6a8:0x3ffc9890 0x40085c61:0x3ffc98d0 0x400d7a25:0x3ffc98f0 0x400d8091:0x3ffc9910 0x400d8fd6:0x3ffc9940 0x400d9091:0x3ffc9970 0x400d9b8b:0x3ffc9990 0x400d5ec0:0x3ffc99b0 0x400eeb4d:0x3ffc99d0 0x4008dc49:0x3ffc99f0

Rebooting...
========================================================

Any idea whay does it happen?
I thought it could be related to not using BLE, given that I'm actually using a Bluetooth 3.0 keyboard...
But not sure as I don't have a BLE (bluetooth 4.0) keyboard here to test it.

Thanks and Best Regards,
Rodrigo Corbera.

kava60, 2020-06-21:
I'm having this same problem. I don't think the unhandled event has anything to do with it however. The panic indicates that an interrupt routine is taking longer than expected to complete resulting in the watchdog timer expiring either because there is a lot of work going on in the interrupt routine, or some sort of tight loop has occurred. In my installation, the interrupt WDT is 300 ms. I'm running on a SparkFun ESP32 Thing which has a dual core at up to 240 MHz but the xtal for the ESP32 is 26 MHz vs 40 MHz for many other ESP32 boards (including the author of this code). As a result I had to recalculate the APLL scaling values to get the proper 14 MHz NTSC frequency (thanks to the author for leaving the necessary code commented out to calculate it). Perhaps the board I'm using is just a little slower than the author's and it is exceeding the 300 ms interrupt watchdog in normal operation or perhaps something about the particular bluetooth keyboard we are using is triggering a bug in the interrupt handler. I'm going to try disabling the interrupt watchdog and see if that helps. Unfortunately it isn't exactly straightforward to do that.

RoCorbera, 2020-06-21:
Thanks for sharing these comments, Kava60.

I tried to debug it further and I got the impression ESP32 reboots when it is about to retrieve Keyboard's device name using Bluetooth Classic protocol... Maybe a pointer problem? Maybe the buffer for the name of the device is too short. Or MTU or other field especification has not enought bytes to handle it?

Other possibility... bluetooth pairing. When I use this keyboard with PC it asks me to type a number of 6 digits as authentication before pairing the device.... This procedure may not be handled by the code and it may cause time out or other issue as well.

I got some panic reboot with ESP32 when going out of memory in some other projects... Panic can be realeted to several possible issues.

ojaksch, 2020-07-20:
Similar issue with my Bluetooth 3.0 keyboard and this [esp32 lolin32](https://azdelivery.de/en/products/esp32-lolin-lolin32?_pos=1&_sid=97fdf1935&_ss=r#description) I bought.
Core is working fine, but when keyboard is in pairing mode while esp32 boots, it keeps crashing/rebooting until keyboard stops it's pairing mode.

RoCorbera, 2020-07-20:
I've tried connecting two different Bluetooth 3.0 keyboards and both fail. 
I also isolated the Bluetooth code of this project to test it, but it fails in the same way.

As a work around, I have tried to use BLE keyboard by adding BLE code to the project, but there isn't enough RAM to run BLE and any emulator (NES, SMS, ATARI) together in the same ESP32.

My final solution uses two ESP32. The first takes care of BLE Keyboard and Gamepads along with wired controls in a separated code. This ESP32 also works as I2C slave control's informing input state. The second ESP32 has I2C Master code and inquires the first ESP32, along with game emulation. 
It works fine and opens the way to think about using BLE/BT alternative modules.

ojaksch, 2020-09-10:
@RoCorbera Do you have any links, code and/or examples for your solution?

smutek110, 2020-12-21:
Similar problem with the esp32 wroom board,reset while pairing. I have already bought three keyboards and one game pad ,
but they all give the same error. In the monitor you can see that the input device has been found - you can see the device address and restart. Over and over again.

derkomai, 2020-12-21:
> Similar problem with the esp32 wroom board,reset while pairing. I have already bought three keyboards and one game pad ,
> but they all give the same error. In the monitor you can see that the input device has been found - you can see the device address and restart. Over and over again.

Same problem here.

closed 2021-06-17 by RoCorbera

AnzuTeam, 2022-01-01:
if anybody still cares, I solved the crash by using bluetooth classic in the bluetooth config.
in hci_transport_esp32.cpp, add this function:
#define BT_MODE ESP_BT_MODE_CLASSIC_BT
bool startBT()
{
    esp_bt_controller_config_t cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    cfg.mode = BT_MODE;
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED){
        return true;
    }
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){
        esp_bt_controller_init(&cfg);
        while(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_IDLE){}
    }
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_INITED){
        int ret = esp_bt_controller_enable(BT_MODE);
        if (ret) {
            printf("BT Enable failed\n");
            return false;
        }
    }
    if(esp_bt_controller_get_status() == ESP_BT_CONTROLLER_STATUS_ENABLED){
        return true;
    }
    printf("BT Start failed\n");
    return false;
}

and call startBT() instead of btStart() in hci_open.

ivorss, 2022-03-12:
I implemented and used the startBT() function from above, however it still crashes:

`
HCI_COMMAND_COMPLETE_EVT 4 bytes
 -> HCI_RESET 0C03 No Error
>HCI_COMMAND_COMPLETE_EVT 11 bytes
 -> HCI_READ_BUFFER_SIZE 1005 No Error
HCI_COMMAND_COMPLETE_EVT 10 bytes
 -> HCI_READ_BD_ADDR 1009 No Error
HCI_COMMAND_COMPLETE_EVT 4 bytes
 -> HCI_WRITE_CLASS_OF_DEVICE 0C24 No Error
HCI_COMMAND_COMPLETE_EVT 4 bytes
 -> HCI_WRITE_SCAN_ENABLE 0C1A No Error
nofrendo inserting /sd/nofrendo/chase.nes
HCI_COMMAND_STATUS_EVT 4 bytes
 -> HCI_INQUIRY 0401 No Error
HCI_COMMAND_COMPLETE_EVT 4 bytes
 -> HCI_WRITE_AUTH_ENABLE 0C20 No Error
nofrendo /sd/nofrendo/chase.nes is 24592 bytes
main: partition type = 64.
main: partition subtype = 64.
main: partition starting address = 300000.
main: partition size = 20000.
main: partition label = rom.
main: partition encrypted = 0.

00010000 00006010 /sd/nofrendo/chase.nes
CrapFS::mmap mapping /sd/nofrendo/chase.nes offset:00010000 len:24592
CrapFS::mmap mapped to 3F4C0000
Head: 0x3f4c0000 (4e 45 53 1a)
HCI_INQUIRY_RESULT_EVT 15 bytes
20:17:04:00:00:B4:402500 input device added
HCI_INQUIRY_RESULT_EVT 15 bytes
HCI_INQUIRY_RESULT_EVT 15 bytes
HCI_INQUIRY_RESULT_EVT 15 bytes
HCI_INQUIRY_RESULT_EVT 15 bytes
HCI_INQUIRY_RESULT_EVT 15 bytes
HCI_INQUIRY_COMP_EVT 1 bytes
HCI_COMMAND_STATUS_EVT 4 bytes
 -> HCI_CREATE_CONNECTION 0405 No Error
HCI_CONNECTION_COMP_EVT 11 bytes
20:17:04:00:00:B4 connected on handle 129
HCI_COMMAND_STATUS_EVT 4 bytes
 -> HCI_REMOTE_NAME_REQUEST 0419 No Error
HCI_MAX_SLOTS_CHANGED_EVT 3 bytes
unhandled hci case: 27
ASSERT_ERR(lm_env.local_name), in lm.c at line 79
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008862d  PS      : 0x00060a34  A0      : 0x80125b81  A1      : 0x3f
fdcde0
0x4008862d: r_assert_err at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x3f
fbdbb4
A6      : 0x3ffbdbb4  A7      : 0x00000000  A8      : 0x80088628  A9      : 0x3f
fdcdc0
A10     : 0x00000032  A11     : 0x00000032  A12     : 0x00000010  A13     : 0xff
ffffff
A14     : 0x00000000  A15     : 0xfffffffd  SAR     : 0x00000004  EXCCAUSE: 0x00
000005
EXCVADDR: 0x00000000  LBEG    : 0x400885d0  LEND    : 0x400885d7  LCOUNT  : 0x00
000000
0x400885d0: r_assert_err at ??:?

0x400885d7: r_assert_err at ??:?

ELF file SHA256: c4cdf6f3c3819b24

Backtrace: 0x4008862d:0x3ffdcde0 0x40125b7e:0x3ffdce00 0x4002509e:0x3ffdce20 0x4
011c09a:0x3ffdce60 0x400895a2:0x3ffdce80 0x40019d11:0x3ffdceb0 0x40055b4d:0x3ffd
ced0 0x40117a7f:0x3ffdcef0 0x40118091:0x3ffdcf10
0x4008862d: r_assert_err at ??:?

0x40125b7e: r_LM_GetLocalNameSeg at ??:?

0x4011c09a: lc_lmp_rx_handler at ??:?

0x400895a2: ke_task_schedule at ke_task.c:?

0x40117a7f: r_rw_schedule at ??:?

0x40118091: btdm_controller_task at ??:?

Core 1 register dump:
PC      : 0x401a53e6  PS      : 0x00060534  A0      : 0x800d21fe  A1      : 0x3f
fce580
0x401a53e6: esp_pm_impl_waiti at C:/msys32/home/ivors/esp/esp-idf/components/esp
32/pm_esp32.c:492

A2      : 0x00000000  A3      : 0x00060023  A4      : 0x00060020  A5      : 0x3f
fcf4c0
A6      : 0x3ffc4b20  A7      : 0x3ffcf5cc  A8      : 0x800d0fb2  A9      : 0x3f
fce550
A10     : 0x00000000  A11     : 0x3ffbdc18  A12     : 0x800920d9  A13     : 0x3f
fd95e0
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00
000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00
000000

ELF file SHA256: c4cdf6f3c3819b24

Backtrace: 0x401a53e6:0x3ffce580 0x400d21fb:0x3ffce5a0 0x40091515:0x3ffce5c0
0x401a53e6: esp_pm_impl_waiti at C:/msys32/home/ivors/esp/esp-idf/components/esp
32/pm_esp32.c:492

0x400d21fb: esp_vApplicationIdleHook at C:/msys32/home/ivors/esp/esp-idf/compone
nts/esp32/freertos_hooks.c:86

0x40091515: prvIdleTask at C:/msys32/home/ivors/esp/esp-idf/components/freertos/
tasks.c:4409

Rebooting...

`