rossumur/esp_8_bit #13: Picture of wiring?

kevs-oc, 2020-06-14:
I was hoping you could post a picture of the actual wiring. I am a beginner at interpreting wiring diagrams but was hoping to build this for my son's gradution.

Thanks!

CirclesCenter, 2020-06-15:
I can help! The audio cable has a positive and a ground and the capacitor goes between the audio ground and and the ground pin on the ESP32. The positive wire goes to a 1k resistor that by default goes to pin 18. The circuit works by modulating the discharge of the capacitor to generate the audio

I sketched this for you. Let me know if it helps! https://i.ibb.co/WGqHJHQ/ESP8BIT.png

kevs-oc, 2020-06-15:
This is perfect, just what I needed. Thank you so much!

closed 2020-06-15 by kevs-oc