rossumur/esp_8_bit #15: Connecting 2nd esp32 as HID

llamitaiot, 2020-06-21:
I got a snippet from here:
https://github.com/nkolban/esp32-snippets/issues/230#issuecomment-352197634

And although my phone detects it as a keyboard, I can't get it to connect with the esp-8bit.

My plan is to put the esp32 in an old nes controller.

Btw this project is awesome. All the beers for you if you ever come around Argentina once we're done with covid.

RoCorbera, 2020-06-22:
You should have in mind that Esp_8_bit has a HID stack for classic bluetooth.
BLE and Classic Bluetooth are not compatible with each other.

llamitaiot, 2020-06-22:
Ok, I've been trying to replace the his stuff. But I can't find where in the code I can submit a pressed key to the emulator.
For instance, could you tell me where I can add a line so that as soon a a game boots a start button gets pressed?

llamitaiot, 2020-06-22:
Or maybe a simple gist to make an esp connect as a keyboard using that Bluetooth classic stack. I'd love to contribute to this project, but I need a little help to get started.

RoCorbera, 2020-06-22:
I'm not the author fo this project. But as far as I can see from the code you should work with gui_hid(...) function which in turn handles Keyboard HID events for the console menu (that one used to change games) and also sends controller events to the current emulator by calling _emu->key(...).

Everything works as HID events from a bluetooth keyboard, even <Start> which is keycode 40 (enter key). 

I hope it helps you!

CornN64, 2020-06-26:
I forked this and added support for a hard wired NES controller which allows you to access the file tree with LEFT+SELECT.
It means that you need to solder in 5 leads from a classic NES controller to make it work thought...
If I get the time I might add in a second controller as well (unless someone beats me to it), it should be simple enough to extend it.

djtrance, 2020-07-03:
hi, I'm trying to add bluetooth joystick support , I have a universar joystick techmaster, tm.stk7005,  when I connect the joystick to the emulator it is enter to the function `hid(InputDevice* d, const uint8_t* data, int len)`
in the switch the device enter here
```
        case 0x07: //test joystick bluetooth tecmaster tm.stk7005
                    {
                    printf("bluetooth tecmaster -> report: %02X\n",h[2]);
                    int sizeB = (h[2] >> 4) + 1;
                    int addrB = (h[3] << 8) | h[4];  // low bytes of address
                    h += 5;

                    int idB = (h[4] << 8) | h[5];
                    printf("tecmaster %04X:%02X read id is %04X\n",sizeB,addrB,idB);
                    dump_report(data,len);
                break;
                    }
```
but I don't know how to extract the data from the joystick. I can see how the len change when I move the stick or I press some buttons.
can any help me?

CornN64, 2020-07-05:
Here is something fresh out of the press...
https://github.com/darthcloud/BlueRetro/tree/v0.2
I have not tested it myself but looks useful to those who wish to keep things wireless :)

paulaxinte, 2023-12-14:
Is there any possibility to integrate wireless connection between 2 esp32 for the emulator via ESP-NOW? If yes, where do I need to enter the code I've written for receiving input?