rossumur/esp_8_bit #18: X3 Gamepad working! Not an issue

NoNamedCat, 2020-07-01:
I get working this type of controller, is very cheap and easy to find in ebay or other online sites:

https://i.ytimg.com/vi/N2BsJA6EAOo/maxresdefault.jpg](url)

To make it work you need to press (with the controller OFF) the B button and then press the Home button (while keeping the B button pressed). This make the controller act as a keyboard. When you press any button this will cause a short keypress. When you release the key it cause anther short keypress but from other key. This force me to modify the code in this way:

I dont know how to add this functionality but i share my implementation:

**gui.cpp**:

```
bool key(int keycode, int pressed, int mods)
    {
        printf("key:%02X %02X %02X\n",keycode,pressed,mods);
        if (pressed && _visible)
            _click = 1;

        if (pressed && keycode == 12) { // F1 - GUI key
            _visible = !_visible;       // toggle GUi
            if (_visible)
                _overlay->frame();      // draw the frame when it first appears
            _click = 1;
            return true;
        }
        if (!_visible)
            return false;

        if (pressed) {
            switch (keycode) {
                case 30:        // 1 key
                case 31:        // 2 key
                case 14:        // 0 key
                    disk_key(keycode-30);
                    break;
                case 15:
                    enter(mods);    // return
                    break;
                case 26:    // up
                    move_v(-1);
                    break;
                case 7:    // down
                    move_v(1);
                    break;
                case 27:    // left
                    move_tab(-1);
                    break;
                case 4:    // right
                    move_tab(1);
                    break;
                case 11:   // left control key
                    break;  // FIRE
            }
        }
        return true;
    }
```

**emu_nofrendo.cpp**

```
 virtual void key(int keycode, int pressed, int mods)
    {
        switch (keycode) {
            case 26: pad(1,event_joypad1_up); break;
            case 8: pad(0,event_joypad1_up); break;

            case 7: pad(1,event_joypad1_down); break;
            case 6: pad(0,event_joypad1_down); break;

            case 27: pad(1,event_joypad1_left); break;
            case 29: pad(0,event_joypad1_left); break;

            case 04: pad(1,event_joypad1_right); break;
            case 20: pad(0,event_joypad1_right); break;

            case 47: pad(1,event_soft_reset); break; // soft reset - 'r'
            case 48: pad(0,event_soft_reset); break; // soft reset - 'r'

            case 30: pad(1,event_hard_reset); break; // hard reset - 't'
            case 31: pad(0,event_hard_reset); break; // hard reset - 't'

            case 15: pad(1,event_joypad1_start); break; // F4
            case 25: pad(0,event_joypad1_start); break; // F4

            case 18: pad(1,event_joypad1_select); break; // tab
            case 10: pad(0,event_joypad1_select); break; // tab

            case 11: pad(1,event_joypad1_a); break; // left shift key event_joypad1_a
            case 21: pad(0,event_joypad1_a); break; // left shift key event_joypad1_a

            case 24: pad(1,event_joypad1_b); break; // option key event_joypad1_b
            case 9: pad(0,event_joypad1_b); break; // option key event_joypad1_b

            //case 59: system(pressed,INPUT_PAUSE); break; // F2
            //case 61: system(pressed,INPUT_START); break; // F4
            //case 62: system(pressed,((KEY_MOD_LSHIFT|KEY_MOD_RSHIFT) & mods) ? INPUT_HARD_RESET : INPUT_SOFT_RESET); break; // F5
        }
    }
```
how we can add this functionality?

RoCorbera, 2020-07-06:
Hello, It's interesting that this gamepad works fine. 
Have you tested it with a game on the ESP8Bit? 

If you keep the pad pressed to any direction, does it keep moving the "player" in the screen?
I ask that because from my experience a simple "short keypress" moves it a bit only and it doesn't keep moving the "player" continuosly in the game.

Would you please share a video with tihs gamepad working with ESP8Bit?

Thanks.

vladimir-kuznetsov, 2021-03-11:
@NoNamedCat 
> I get working this type of controller, is very cheap and easy to find in ebay or other online sites:

Hi, I've bought X3 gamepad and trying to use it with ESP32 and this awesome project.
Have you figure it out how to pair X3?
When I trying to connect it (in any mode B/X/Y/A) things always ends up with message:
"pin didn't work, won't create a link key: 1073523896"
After that it switches to another mode and start sending ACL messages with a code of pressed key:
"< ACL 812010000C004200A10780808080881000000000"
It's possible to process this but I think it should be better to pair devices.

NoNamedCat, 2021-04-16:
Strange behavior... Maybe the gamepad had another firmware? Maybe it has another micro? The code that I posted have been working for me to these days

vladimir-kuznetsov, 2021-04-16:
> Strange behavior... Maybe the gamepad had another firmware? Maybe it has another micro? The code that I posted have been working for me to these days

I think it might be another revision of the firmware but it still uses the same cheap Chinese chip (ShanWan). It pairs easily with Android and I dumped a connection log. It uses Secure Simple Pairing instead of a legacy pairing pin. But after rejecting the pairing pin it starts working without pairing. So I made my hacky fake HID handling based on ACL messages and even have the ability to handle analog values.
But it still would be better to implement SSP.