rossumur/esp_8_bit #21: Running this on esp32-s2?

bootrino, 2020-07-19:
I'd like to know if this would run on an esp32-s2.

If any of the project contributors are interested in trying, I'd be happy to pay for a couple of esp32-s2 boards for you to use. 

I know it's not much money to spend but it's not really fair to ask without volunteering the hardware too.

If any of the project devs are interested, let me know and we'll work out how I can pay for the purchase.

RoCorbera, 2020-07-20:
Does S2 support USB Host interface and protocol? 
I don't think so. As far as I have read, it only supports Client USB...

bootrino, 2020-07-21:
@RoCorbera I'm pretty sure they are working on host mode.

See thread:

https://www.esp32.com/viewtopic.php?t=16147

RoCorbera, 2020-07-21:
We can follow the specs at https://esp32.com/viewtopic.php?f=10&t=14532
So far:
 USB Device — Supported
 USB Host — Not supported (IDF-530)

and based on the status of this issue (https://github.com/espressif/esp-idf/issues/5454):
Ivan Grokhotkov (from Epressif/Shangai) answered to Chegewara (support from Espressif): "Hi @chegewara, unfortunately we don't have timeline for the USB Host feature, yet." on Jun, 15th.

Thus I have not much hope for it...

chegewara, 2021-01-09:
Support for USB host will be added for sure on S2, especially that S3 is on its way which also support native USB. The problem is it is in very early stage in tinyUSB library.

RoCorbera, 2021-01-11:
Thanks for the update, @chegewara !
Would it work for ESP32 WROOM as well?

For future reference of others, here are some links about tinyUSB and ESP32:
https://github.com/hathach/tinyusb
https://www.esp32.com/viewtopic.php?t=15908
https://github.com/chegewara/esp-idf/tree/master/examples/peripherals/usb
https://github.com/chegewara/esp32s2-tinyusb-arduino-as-component

chegewara, 2021-01-11:
> Would it work for ESP32 WROOM as well?

No, esp32 does not support native USB.

btw, there is missing 1 library from my repo ;)
https://github.com/chegewara/EspTinyUSB