rossumur/esp_8_bit #28: REQUEST: Binary image

Zibri, 2020-08-14:
Can you release also a binary image to flash with Flash Download Tool?

sansaloneverretn, 2020-09-23:
[4Mb-esp32-8-bit.zip](https://github.com/rossumur/esp_8_bit/files/5271750/4Mb-esp32-8-bit.zip)
esptool.py --port COM0 write_flash --flash_mode dio --flash_size 4MB 0x0 4Mb-esp32-8-bit.bin

Zibri, 2020-09-24:
ohhhh thanks!!!!

Zibri, 2020-09-24:
is there an ATARI 2600 version? I would love that

Zibri, 2021-07-04:
> NES version: [4Mb-esp32-8-bit.zip](https://github.com/rossumur/esp_8_bit/files/5271750/4Mb-esp32-8-bit.zip)
> ATARI version: [4Mb-esp32-8-bit-atari.zip](https://github.com/rossumur/esp_8_bit/files/5271934/4Mb-esp32-8-bit-atari.zip)
> video pin: 25
> audio pin: 26
> esptool.py --port COM0 write_flash --flash_mode dio --flash_size 4MB 0x0 4Mb-esp32-8-bit.bin

just to questions: how did you create the fikles on github? I can't find the in the repository.
I don't want to bother you in the future to check for updates.
thanks again @sansaloneverretn @rossumur

reppad, 2021-08-19:
Each of the 2 files is a full dump of the 4MB ESP flash memory so you have to unzip it and flash bin file at address 0x0, for example with esptool :
`esptool.py -b 921600 -p /dev/ttyUSB0 write_flash 0x0 /path/to/4Mb-esp32-8-bit.bin`