rossumur/esp_8_bit #29: replace bt-stack with bluekitchen

sansaloneverretn, 2020-09-23:
Is there any chance to replace bt-stack with https://github.com/bluekitchen/btstack ???
https://gitlab.com/ricardoquesada/unijoysticle2/blob/master/docs/supported_gamepads.md

please rewrite video_out.h for esp-idf framework.

bootrino, 2020-09-23:
The problem with bluetooth is memory usage.

My understanding is this project implements the minimum bluetooth code needed to support the controllers it supports.

sansaloneverretn, 2020-09-23:
ok, what about another bt-stack like: https://hackaday.io/project/170365-blueretro/log/178249-learning-bluetooth-classic-bredr-with-hci-traces from WiFi Linux driver developer ???
**Jacques Gagnon**: For BlueRetro I decided to write my own "minimal" stack talking to the BT controller directly via the ESP32's VHCI interface. My reasoning for doing so was that I knew nothing about Bluetooth and duplicating what I was seeing in the HCI traces directly to the controller looked easier and faster than learning a particular stack API...
https://github.com/darthcloud/BlueRetro

bootrino, 2020-10-01:
I suspect you'll have to write the code yourself if you want this.