rossumur/esp_8_bit #30: Can I use NES Controller? How to set it up? Is there a code for the relevant input device?

38293312, 2020-10-07:
Can I use NES Controller? How to set it up? Is there a code for the relevant input device? Thanks！

sansaloneverretn, 2020-10-07:
https://github.com/rossumur/esp_8_bit/pull/16/commits/4ecbfb7ab257447aac2f11abceff6aada4d5e9fe

38293312, 2020-10-08:
> [4ecbfb7](https://github.com/rossumur/esp_8_bit/commit/4ecbfb7ab257447aac2f11abceff6aada4d5e9fe)

Thank you, I just learned ESP32, if I don’t understand, I will consult you again!

38293312, 2020-10-08:
> [4ecbfb7](https://github.com/rossumur/esp_8_bit/commit/4ecbfb7ab257447aac2f11abceff6aada4d5e9fe)

Thank you very much, now you can play the game normally. Hahaha! !

manogyapro, 2025-12-07:
https://github.com/rossumur/esp_8_bit/issues/30#issuecomment-705253804
hello could you help me also, could you please help em to control using physical button(pushbutton).