rossumur/esp_8_bit #31: Should this work RCA to AV cable? Getting video noise

derkomai, 2020-12-13:
I'm trying to make the sketch work in a ESP32 mini, and since I do not have any composite video-capable devices at home, I'm trying a 3xRCA to AV cable. After a few seconds connected, I can see some noisy vertical lines and some noisy colors from time to time, but no sound. Not sure if it's a connection problem of if there is something I'm missing. Maybe I didn't understand the scheme.

What I've done:
- Solder positive of video cable to pin 25
- Solder negative of video cable to ground
- Solder both audio left and right positives to pin 26 through a 1k resistor (and set pin 26 as audio out in video_out.h)
- Solder both audio left and right negatives to ground through a 10nf capacitor

[image]

cziter15, 2020-12-13:
Please note that these signals should be 75Ohm impedance matched. Do you
have all shieldings connected to gnd?

Try this:
To pin 25 - positive video (yellow), outer/shield directly to GND
To pin 26 - throug capacitor 10uF to left and right positive audio,
outer/shield directly to GND

Regards,
Chris

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Wolny
od wirusów. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

derkomai, 2020-12-13:
@cziter15, I read [here](https://www.reddit.com/r/esp32/comments/gosbxg/esp_8_bit_atari_8_bit_computers_nes_and_sms_game/) the following comment:

> If you're having the same problem as me the capacitor goes on the negative of the audio line, not T'd off the positive like it seems to newbs.

That's why I soldered the audio L/R outter/negative/shielding through the capacitor to ground. So should I solder the capacitor to the positive then?

What about the resistor? The scheme lists a 1k resistor in the audio line. Should I remove it? I'm a little confused about the 750 ohm impedance matching. 

```
    -----------
    |         |
    |      25 |------------------> video out
    |         |
    |      18 |---/\/\/\/----|---> audio out
    |         |     1k       |
    |         |             ---
    |  ESP32  |             --- 10nf
    |         |              |
    |         |              v gnd
    |         |
    |         |     3.3v <--+-+   IR Receiver
    |         |      gnd <--|  )  TSOP4838 etc.
    |       0 |-------------+-+   (Optional)
    -----------
```

This is how I have it:
```
Video positive/yellow cable -----------------------------------PIN 25
Video negative/outter cable ---------------------------------- GND
Audo L/R positives (red & white) -------- 1k resistor ---------PIN 26
Audo L/R negatives/outter ---------- 10 nF capacitor ----------GND
```

Martin-Laclaustra, 2020-12-13:
Verify that you are using the correct connections for your device on the other end.
TRRS is far from being standardized:
https://en.wikipedia.org/wiki/Phone_connector_(audio)#TRRS_standards

derkomai, 2020-12-14:
> Verify that you are using the correct connections for your device on the other end.

Interesting. How should I do that? The device is a cheap projector that has no instructions manual. Concerning the connections I listed on my last message, do you feel they're correct?

derkomai, 2020-12-14:
I've tried with another ESP32 on a breadboard. I've tried plugging pin 25 to every of the 3 different lines and none seems to output a video signal. Surely I must be doing something very wrong.

derkomai, 2020-12-14:
I can confirm that [Bitluni CompositeVideo](https://github.com/bitluni/ESP32CompositeVideo) repository works with the actual cabling, and I can see the composite video demo. Still, esp_8_bit does not output video nor audio yet.

CornN64, 2020-12-22:
You can not use pin 26 for audio!
In fact both pin 25 & 26 outputs video and can't be changed to anything else.
You'll need to use some other pin like the one suggested pin 18