rossumur/esp_8_bit #32: Roms not showing up in the menu.

markoni985, 2020-12-27:
After placing the ROMs in the data folder for the appropriate emulator not all of them show up in the menu.
I kept the names short, with no spaces. Sometimes if I change names device ends in a boot loop where it tries to format FS. A list of supported ROMs and appropriate names would go a long way.

CornN64, 2020-12-31:
I would say that most ROMs works only some exotic ones will fail.
You only have about 1Mbyte of space to store them (in the internal flash memory) so in principle you can only have a handful of ROMs unless you use a SD card for storage (https://github.com/CornN64/esp_8_bit).

markoni985, 2021-01-07:
I'll give that a try when once sd card module arrives.
I tried deleting SPIFFS using : 

#include "SPIFFS.h"
 
void listAllFiles(){
 
  File root = SPIFFS.open("/");
 
  File file = root.openNextFile();
 
  while(file){
 
      Serial.print("FILE: ");
      Serial.println(file.name());
 
      file = root.openNextFile();
  }
 
}
 
void setup() {
 
  Serial.begin(115200);
 
  if (!SPIFFS.begin(true)) {
    Serial.println("An Error has occurred while mounting SPIFFS");
    return;
  }
 
  Serial.println("\n\n----Listing files before format----");
  listAllFiles();
 
  bool formatted = SPIFFS.format();
 
  if(formatted){
    Serial.println("\n\nSuccess formatting");
  }else{
    Serial.println("\n\nError formatting");
  }
 
  Serial.println("\n\n----Listing files after format----");
  listAllFiles();
}
 
void loop() {}

I also tried compiling and loading on another computer to make sure there are no hidden temp files affecting the compile process.

And I am still getting:

esp_8_bit

mounting spiffs (will take ~15 seconds if formatting for the first time)....
... mounted in 99 ms
frame_time:0 drawn:1 displayed:0 blit_ticks:0->0, isr time:0.00%
emu_task nofrendo running on core 0 at 240000000mhz
nofrendo inserting /nofrendo/chase.nes
nofrendo /nofrendo/chase.nes is 24592 bytes
main: partition type = 0.
main: partition subtype = 17.
main: partition starting address = 150000.
main: partition size = 140000.
main: partition label = app1.
main: partition encrypted = 0.

00010000 00004010 /nofrendo/tokumaru_raycast.nes
00020000 00020000 /smsplus/baraburuu.sms
00040000 00010000 /smsplus/ftrack.gg
00050000 0000A010 /nofrendo/sokoban.nes
00060000 0005BB00 /nofrendo/smb3.nes
000C0000 00010010 /nofrendo/adventureisland.nes
000E0000 00010010 /nofrendo/ai.nes
00100000 00020010 /nofrendo/contra .nes
00130000 00040010 /nofrendo/cf.nes
00180000 00006010 /nofrendo/chase.nes
00190000 00006010 /nofrendo/mariobros.nes
001A0000 0000FB00 /nofrendo/marioyoshi.nes
001B0000 00010010 /nofrendo/advild.nes
001D0000 00015900 /nofrendo/smb2.nes
001F0000 00010010 /nofrendo/drmario.nes
00210000 0000608F /nofrendo/bc.nes
00220000 00040000 /smsplus/sonic.sms
00260000 00004010 /nofrendo/tokumaru_raycast.nes
CrapFS::mmap mapping /nofrendo/chase.nes offset:00180000 len:24592
nofrendo can't map /nofrendo/chase.nes
frame_time:3 drawn:141 displayed:0 blit_ticks:-1->0, isr time:0.00%
frame_time:3 drawn:1495 displayed:0 blit_ticks:-1->0, isr time:0.00%
frame_time:3 drawn:2769 displayed:0 blit_ticks:-1->0, isr time:0.00%
frame_time:3 drawn:4048 displayed:0 blit_ticks:-1->0, isr time:0.00%
frame_time:3 drawn:5316 displayed:0 blit_ticks:-1->0, isr time:0.00%
frame_time:3 drawn:6590 displayed:0 blit_ticks:-1->0, isr time:0.00%
frame_time:3 drawn:7865 displayed:0 blit_ticks:-1->0, isr time:0.00%

These games are not supposed to be there.
Any idea where is this file structure kept so I can delete it?
I have a feeling that the script deletes files but not the file structure.
Any way to zero-fill SPIFFS?

Thank you.

Thank you in advance.

CornN64, 2021-01-07:
One way would be to use the download tool
Get the flash_download_tool_v3.8.5 (https://www.espressif.com/en/support/download/other-tools)
Hit developer mode -> ESP32 downloadtool
Make sure correct com port is selected and hit ERASE button, that will wipe everything on the ESP32

markoni985, 2021-01-07:
Awesome, that worked.
Thank you so much.