rossumur/esp_8_bit #34: When compiling the fork esp_8_bit / CornN64 with the EMU_SMS emulator set, he gets an error:

smutek110, 2021-04-20:
c: /users/jessica/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin /../ lib / gcc / xtensa- esp32-elf / 5.2.0 /../../../../ xtensa-esp32-elf / bin / ld.exe: C: \ Users \ Jessica \ AppData \ Local \ Temp \ arduino-sketch-C8DA85705841A54D78442418212C2586 /esp_8_bit.ino.elf section `.dram0.bss 'will not fit in region` dram0_0_seg'
c: /users/jessica/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin /../ lib / gcc / xtensa- esp32-elf / 5.2.0 /../../../../ xtensa-esp32-elf / bin / ld.exe: DRAM segment data does not fit.
c: /users/jessica/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin /../ lib / gcc / xtensa- esp32-elf / 5.2.0 /../../../../ xtensa-esp32-elf / bin / ld.exe: region `dram0_0_seg 'overflowed by 512 bytes
collect2.exe: error: ld returned 1 exit status
Compilation error: Error: 2 UNKNOWN: exit status 1

CornN64, 2021-04-21:
I am not sure whats up but it looks like it can not fit the code in DRAM. I do not get this problem and I am using Arduino 1.8.13 version to compile and upload the binary.
Would be interesting to know if anyone else has this issue...

smutek110, 2021-04-22:
I am using Arduino 1.8.14  also tried the version 2.0.0beta  i use esp32 wroom. The compile error is only for emu_sms. Will try with other esp32 boards. I will also replace flash with winbound 25q128fvsg and I will write if it changes anything

smutek110, 2021-04-22:
73 / 5000
Wyniki tłumaczenia
It will also try to install arduino and libraries on a clean system

smutek110, 2021-04-22:
It works fine on a clean system, apparently it's a matter of libraries and interdependent files. Sorry for the confusion and thank you for your time. I'm just gaining the knowledge I need, and I'm a self-taught person

CornN64, 2021-04-22:
I am happy you got it sorted out

Cheers

closed 2021-04-23 by smutek110

CornN64, 2021-05-27:
So just to finish this...
As far as I can tell there seems to be an issue with board manger esp32 library newer than 1.0.4 (namely 1.0.5 and 1.0.6)
As soon as I upgrade to these newer versions and try to compile SMS (the other emulators seems fine) I get the error

esp_8_bit.ino.elf section .dram0.bss 'will not fit in region dram0_0_seg'
ld.exe: DRAM segment data does not fit.
ld.exe: region `dram0_0_seg 'overflowed by 512 bytes

Maybe someone can enlighten us to why this happens...