rossumur/esp_8_bit #38: Compilation error

the-red-eye-studio, 2021-08-18:
The error:
```
In file included from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:9:
C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\video_out.h:286: warning: "IRE" redefined
 #define IRE(_x)          ((uint32_t)(((_x)+40)*255/3.3/147.5) << 8)   // 3.3V DAC
 
In file included from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:43: note: this is the location of the previous definition
 #define IRE(_x)          ((uint32_t)((_x+40)*255/3.3/147.5) << 8)   // 3.3V DAC
 
In file included from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:9:
C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\video_out.h:290: warning: "GRAY_LEVEL" redefined
 #define GRAY_LEVEL       IRE(50)
 
In file included from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:47: note: this is the location of the previous definition
 #define GRAY_LEVEL       IRE((100-7.5)/2)
 
In file included from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:25,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
c:\users\the red eye studio\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\sys-include\dirent.h:53:6: error: conflicting declaration of 'DIR* opendir(const char*)' with 'C' linkage
 DIR *opendir(const char *);
      ^~~~~~~
In file included from c:\users\the red eye studio\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\sys-include\dirent.h:39,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:25,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
C:\Users\The red eye Studio\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/sdk/esp32/include/newlib/platform_include/sys/dirent.h:53:6: note: previous declaration with 'C++' linkage
 DIR* opendir(const char* name);
      ^~~~~~~
In file included from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:25,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
c:\users\the red eye studio\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\sys-include\dirent.h:56:3: error: conflicting declaration of 'dirent* readdir(DIR*)' with 'C' linkage
   readdir(DIR *);
   ^~~~~~~
In file included from c:\users\the red eye studio\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\sys-include\dirent.h:39,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:25,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
C:\Users\The red eye Studio\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/sdk/esp32/include/newlib/platform_include/sys/dirent.h:54:16: note: previous declaration with 'C++' linkage
 struct dirent* readdir(DIR* pdir);
                ^~~~~~~
In file included from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:25,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
c:\users\the red eye studio\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\sys-include\dirent.h:58:6: error: conflicting declaration of 'int readdir_r(DIR*, dirent*, dirent**)' with 'C' linkage
 int  readdir_r(DIR *__restrict, struct dirent *__restrict,
      ^~~~~~~~~
In file included from c:\users\the red eye studio\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\sys-include\dirent.h:39,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:25,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
C:\Users\The red eye Studio\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/sdk/esp32/include/newlib/platform_include/sys/dirent.h:59:5: note: previous declaration with 'C++' linkage
 int readdir_r(DIR* pdir, struct dirent* entry, struct dirent** out_dirent);
     ^~~~~~~~~
In file included from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:25,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
c:\users\the red eye studio\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\sys-include\dirent.h:61:7: error: conflicting declaration of 'void rewinddir(DIR*)' with 'C' linkage
 void  rewinddir(DIR *);
       ^~~~~~~~~
In file included from c:\users\the red eye studio\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\sys-include\dirent.h:39,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:25,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
C:\Users\The red eye Studio\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0-rc1/tools/sdk/esp32/include/newlib/platform_include/sys/dirent.h:57:6: note: previous declaration with 'C++' linkage
 void rewinddir(DIR* pdir);
      ^~~~~~~~~
In file included from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:25,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
c:\users\the red eye studio\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\sys-include\dirent.h:72:7: error: conflicting declaration of 'void seekdir(DIR*, long int)' with 'C' linkage
 void  seekdir(DIR *, long);
       ^~~~~~~
In file included from c:\users\the red eye studio\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r1\xtensa-esp32-elf\sys-include\dirent.h:39,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\src\emu.h:25,
                 from C:\Users\The red eye Studio\Desktop\esp_8_bit-master\esp_8_bit\esp_8_bit.ino:8:
[... 195 lines cut ...]
    if (_lines) {
      printf("video_init\n");
      video_init(_emu->cc_width,_emu->flavor,_emu->composite_palette(),_emu->standard); // start the A/V pump
      _inited = true;
    } else {
      vTaskDelay(1);
    }
  }
  #endif
  
  // update the bluetooth edr/hid stack
  hid_update();

  // Dump some stats
  perf();
}
```

The Folder:

[image]

The Rom:

[image]

reppad, 2021-08-19:
Same for me :/

CornN64, 2021-09-19:
Did you try to compile it with Arduino 1.8.13 and the 1.0.4 esp32 board manager lib?

pjh64, 2022-02-03:
Ran into the same issue and can confirm that it works with the 1.0.4 esp32 board manager lib. The m5stack lib I had installed before didn't. Thanks, CornN64!

closed 2022-06-21 by the-red-eye-studio