rossumur/esp_8_bit #42: Reboot after keyboard connect

retro333, 2022-01-27:
My bt keyboard connect after video_init.
The console content:

video_init
FF:FF:FF:FF:FF:FF:402500 input device added
frame_time:6121 drawn:121 displayed:130 blit_ticks:2233->2782, isr time:27.33%
frame_time:6155 drawn:241 displayed:250 blit_ticks:2232->2786, isr time:25.16%
s:64 L2CAP_OPENING
ASSERT_ERR(lm_env.local_name), in lm.c at line 79
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008464c  PS      : 0x00060534  A0      : 0x801163dd  A1      : 0x3ffce100  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x60008048  A5      : 0x3ffbdbb4  
A6      : 0x3ffbdbb4  A7      : 0x00000000  A8      : 0x8008464c  A9      : 0x3ffce0e0  
A10     : 0x00000032  A11     : 0x00000032  A12     : 0x00000010  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffd  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400845f4  LEND    : 0x400845fb  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4008464c:0x3ffce100 0x401163da:0x3ffce120 0x4002509e:0x3ffce140 0x4010c892:0x3ffce180 0x400855ca:0x3ffce1a0 0x40019d11:0x3ffce1d0 0x40055b4d:0x3ffce1f0 0x4010825b:0x3ffce210 0x40108869:0x3ffce230 0x40090756:0x3ffce260

Core 1 register dump:
PC      : 0x400ecfa6  PS      : 0x00060134  A0      : 0x800ece10  A1      : 0x3ffc9e60  
A2      : 0x3ffc6920  A3      : 0x3ffc9e84  A4      : 0x80081832  A5      : 0x3ffc0150  
A6      : 0x00001036  A7      : 0x3ffc235c  A8      : 0x800ecfa6  A9      : 0x3ffc9e40  
A10     : 0x00000000  A11     : 0x3ffc2368  A12     : 0x8008155e  A13     : 0x3ffc0130  
A14     : 0x00000bb4  A15     : 0x3ffe0c40  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400ecfa6:0x3ffc9e60 0x400ece0d:0x3ffc9e80 0x400ecea1:0x3ffc9eb0 0x400ed99f:0x3ffc9ed0 0x400d3190:0x3ffc9ef0 0x400ee819:0x3ffc9f10 0x40090756:0x3ffc9f30

Rebooting...

sdmtr, 2022-02-27:
Same problem as noted in issue #10 (which was closed without a resolution, apparently). The last comment by AnzuTeam provides some alternative code that may fix the problem for you; it didn't for me.

I don't have an infrared keyboard to test with, but I've tried several different bluetooth devices including a Logitech MX Keys for Mac, a Switch Pro Controller, the standard Switch Joycons, a generic cheap bluetooth keyboard, a Rii bluetooth remote, and a couple of other bits and pieces, and all of them caused the same panic.

ivorss, 2022-03-12:
The workaround in #10 did not work for me either.

bigbn, 2022-03-12:
Have the same issue with Xiaomi chinese bluetooth gamepad.  It has a very unusual name, and maybe it is the real reason
of 
`ASSERT_ERR(lm_env.local_name), in lm.c at line 79`
maybe the connection handlers just cant decode them to use properly? Maybe there is any way to include some generic "Stub" for such kind situations?

[image]

ivorss, 2022-03-12:
I don't think the issue is the name, since the BT keyboard I am using has a short English name and I can see it in Windows 10. The crash happens because the connection process halts for some reason and the watchdog gets triggered. I can turn off the watchdog timer from the esp32 configuration and then it doesn't crash and reboot, but the core running BT still hangs.

CornN64, 2022-04-13:
You can try to drop the back-trace info into the Esp Exception Decoder
https://github.com/me-no-dev/EspExceptionDecoder
and see if it helps to trace what went wrong

Cheers

bigbn, 2022-04-28:
This is what i've got:
```
PC: 0x40084f2d
EXCVADDR: 0x00000000

Decoding stack results
0x400920d9: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
```

which after a little googling points to a this: https://github.com/espressif/arduino-esp32/issues/3028

bigbn, 2022-04-28:
Also what I found in google about this place `freertos/port.c:143`:
```
Wifi::event_handler(void*, char const*, int, void*) main/Wifi.cpp:1271 (discriminator 8)
Wifi::eventMarshaller(void*, char const*, int, void*) main/Wifi.cpp:1190
handler_execute at C:/SysGCC/esp32/esp-idf/v4.0/components/esp_event/esp_event.c:145
esp_event_loop_run at C:/SysGCC/esp32/esp-idf/v4.0/components/esp_event/esp_event.c:553 (discriminator 3)
esp_event_loop_run_task at C:/SysGCC/esp32/esp-idf/v4.0/components/esp_event/esp_event.c:115
vPortTaskWrapper at C:/SysGCC/esp32/esp-idf/v4.0/components/freertos/port.c:143
```

Final reference here – line 143 points to this small function:

```
#if CONFIG_FREERTOS_TASK_FUNCTION_WRAPPER
// Wrapper to allow task functions to return (increases stack overhead by 16 bytes)
static void vPortTaskWrapper(TaskFunction_t pxCode, void *pvParameters)
{
  pxCode(pvParameters);
  //FreeRTOS tasks should not return. Log the task name and abort.
  char * pcTaskName = pcTaskGetTaskName(NULL);
  ESP_LOGE(“FreeRTOS”, “FreeRTOS Task “%s” should not return, Aborting now!”, pcTaskName);
  abort();
}
#endif ```

bigbn, 2022-05-01:
Updated from esp core from 1.0.1 to 1.0.6 and got something new
```
0x400810b5: blit(unsigned char*, unsigned short*) at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src/video_out.h line 501
0x400812d9: video_isr(void volatile*) at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src/video_out.h line 599
0x40081455: i2s_intr_handler_video(void*) at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src/video_out.h line 58
0x400912a1: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x40090207: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1536
0x40081735: pthread_mutex_lock_internal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/pthread/pthread.c line 625
PC: 0x400810b5: blit(unsigned char*, unsigned short*) at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src/video_out.h line 501
0x40081766: pthread_mutex_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/pthread/pthread.c line 655
0x400e9265: std::mutex::lock() at c:\users\bigbn\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits/gthr-default.h line 748
0x400e9c05: PacketQ::read(std::vector   >&) at c:\users\bigbn\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/mutex line 377
0x400eabb2: HCI::update() at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src\hid_server/hci_server.cpp line 903
0x400eac49: hci_update() at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src\hid_server/hci_server.cpp line 1434
0x400eb743: hid_update() at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src\hid_server/hid_server.cpp line 1007
0x400d2274: loop() at C:\Users\bigbn\Documents\Arduino\esp_8_bit/esp_8_bit.ino line 199
0x400ec5d9: loopTask(void*) at C:\Users\bigbn\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 23
0x400903f2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
EXCVADDR: 0x00000000

Decoding stack results
0x400810b5: blit(unsigned char*, unsigned short*) at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src/video_out.h line 501
0x400812d9: video_isr(void volatile*) at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src/video_out.h line 599
0x40081455: i2s_intr_handler_video(void*) at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src/video_out.h line 58
0x400912a1: vTaskExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 4274
0x40090207: xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1536
0x40081735: pthread_mutex_lock_internal at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/pthread/pthread.c line 625
0x40081766: pthread_mutex_lock at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/pthread/pthread.c line 655
0x400e9265: std::mutex::lock() at c:\users\bigbn\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\xtensa-esp32-elf\bits/gthr-default.h line 748
0x400e9c05: PacketQ::read(std::vector   >&) at c:\users\bigbn\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/mutex line 377
0x400eabb2: HCI::update() at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src\hid_server/hci_server.cpp line 903
0x400eac49: hci_update() at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src\hid_server/hci_server.cpp line 1434
0x400eb743: hid_update() at C:\Users\bigbn\Documents\Arduino\esp_8_bit\src\hid_server/hid_server.cpp line 1007
0x400d2274: loop() at C:\Users\bigbn\Documents\Arduino\esp_8_bit/esp_8_bit.ino line 199
0x400ec5d9: loopTask(void*) at C:\Users\bigbn\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 23
0x400903f2: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

bigbn, 2022-05-01:
`hci_server.cpp line 903` there is 5 seconds timeout

```
        const auto& cb = *((const cbdata*)data);
        switch (evt) {
            case CALLBACK_READY:
                hci_start_inquiry(5);       // inquire for 5 seconds
                break;
```                
I can increase it, but after specified delay device will still reboot anyway in specified seconds