rossumur/esp_8_bit #45: HID (controller/keyboard) problem with new Arduino ESP32 core 2 ?

NoNamedCat, 2022-03-15:
the HID server is the problem? 
The program works when i comment this line 

`hid_update();`

If I try the emulator with the line uncommented it shows this error on the serial port:

````
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0

esp_8_bit

mounting spiffs (will take ~15 seconds if formatting for the first time)....
... mounted in 132 ms
frame_time:0 drawn:1 displayed:0 blit_ticks:0->0, isr time:0.00%
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e320c: 000b5ee5 e5004136 0a2d0b5d
````

````
Decoding stack results
0x400e30b1: HCI::update() at /home/user/Arduino/arduino-1.8.16/sketchbook/esp_8_bit/src/hid_server/hci_server.cpp line 898
0x400e3111: hci_update() at /home/user/Arduino/arduino-1.8.16/sketchbook/esp_8_bit/src/hid_server/hci_server.cpp line 1434
0x400e38bb: hid_update() at /home/user/Arduino/arduino-1.8.16/sketchbook/esp_8_bit/src/hid_server/hid_server.cpp line 1007
0x400d2668: loop() at /home/user/Arduino/arduino-1.8.16/sketchbook/esp_8_bit/esp_8_bit.ino line 164
0x400e4985: loopTask(void*) at /home/user/Arduino/arduino-1.8.16/packages/esp32/hardware/esp32/2.0.2/cores/esp32/main.cpp line 50
````

NoNamedCat, 2022-03-15:
Also i need to comment this line for the compilation procces
```_emu->name.c_str(),xPortGetCoreID(),rtc_clk_cpu_freq_value(rtc_clk_cpu_freq_get()));```

CornN64, 2022-04-13:
Probably using a newer board manager than version 1.04 ?
If you have the time try to use version 1.04 and see if the problem goes away...

Cheers

NoNamedCat, 2022-05-19:
The problem goes away. What we need to change in the code so it works with the newer version of board manager?