rossumur/esp_8_bit #47: for CornN64 esp_8_bit emu

steevemadison, 2022-08-09:
i have a suggestion for you , if you are willing to merge the coleco and sg1000 in your updated esp_8_bit emu,
i have tried them , and they both work fine

sg1000 emu [https://github.com/EMWiRES/ESP_8_BIT_MINUS](url)
coleco emu [https://github.com/EMWiRES/ESP_8_BIT_PLUS](url)

thank you for your time

CornN64, 2022-08-20:
Nice, I didn't know that someone had worked on these emus, thanks for the heads up.
I cant promise but I'll see if I get the time to add them (unless someone else beat me to it :))

Cheers

magnetus26, 2023-01-02:
I was looking into this but don't know how to integrate them into esp_8_bit. this will be a nice addition

EMWiRES, 2023-02-12:
Be sure to check out the new ESP_8_BIT_PLUS. Better ColecoVision emulation and support for the 8BitDo Zero 2 and FC30 Bluetooth gamepads.