rossumur/esp_8_bit #50: Vertical sync

general-ackbar, 2022-10-26:
I'm having a weird sync problem with one particular screen. It's a dismantled screen from a cheap car rear camera monitor with composite input. The display is not great, but it works fine with other devices (Raspberry Pi, game consoles etc) however with esp_8_bit (or espflix for that matter) the image is crisp but it slowly scrolls vertical (same result with both PAL and NTSC). The output is crisp and steady on other displays so I'm thinking it must be something with this specific combination - and most likely the sync signal passed from the esp32 to the display or maybe the resolution. Sync signals and timing have never been my strong suit, so I haven't been able to find where in the source code the resolution and/or sync can be adjusted. Can anyone point me in the right direction?

CornN64, 2022-11-20:
Have a look in video.h
Each line of video is created there and include the horizontal sync as well as the vertical sync.

scanlight, 2024-01-14:
i have the same issue. video keeps scrolling.

scanlight, 2024-01-15:
Found out that modding the video.h file and reduce  pal and ntsc by one scanline stops the scrolling. 
Now I still need to get a more clear image. It all looks fuzzy with barely color.

general-ackbar, 2024-01-23:
Glad it worked for you :) I found the lines in Corn64 suggestion too, but for my screen I could only make the scrollign go slower or the opposite direction by tweaking the scanlines. For the color and fuzzy-ness problem, that could be ordinary interference. Are you using shielded cables?