rossumur/esp_8_bit #52: Esp32-cam bootlooping with error

mikimause9, 2023-01-04:
Hi, I must say that this project is amazing, and it is a shame that its development was stopped. I hope that somebody could help me anyway.
 I tried to run this software on an ESP32-CAM board, and after solving problems like downgrading the ESP32 board package to 1.0.4, I was able to compile it and upload it to the microcontroller. The problem is that it does not load. These are the logs:
-_-_-_-_-_-_--
esp_8_bit

mounting spiffs (will take ~15 seconds if formatting for the first time)....
... mounted in 6375 ms
frame_time:0 drawn:1 displayed:0 blit_ticks:0->0, isr time:0.00%
emu_task smsplus running on core 0 at 240000000mhz
unpacking /smsplus/ftrack.gg
unpacking /smsplus/baraburuu.sms
unpacking /smsplus/nanowars8k.sms
init_screen
CrapFS::CrapFS app1 not found
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400e5e9f  PS      : 0x00060d30  A0      : 0x800e6141  A1      : 0x3ffe33a0  
A2      : 0x3ffe33d4  A3      : 0x3ffe33fc  A4      : 0x00020000  A5      : 0x00010000  
A6      : 0x00060021  A7      : 0x00000000  A8      : 0x801180ad  A9      : 0x3ffe3380  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000000c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400e5e9f:0x3ffe33a0 0x400e613e:0x3ffe33c0 0x400e6ebe:0x3ffe3450 0x400e7eb9:0x3ffe34a0 0x400e7f40:0x3ffe3500 0x400e8a9d:0x3ffe3520 0x400e8ad9:0x3ffe3580 0x400d21a1:0x3ffe35a0 0x400d224b:0x3ffe35e0 0x40090229:0x3ffe3600

Rebooting...
-_-_-_-_-_-_--

I've tried to track the problem and I found out that this function in /src/gui.cpp creates error:
    void insert(const string& path, int flags)
    {
        set_pref("recent",path);
        _emu->insert(_path + "/" + path,flags);//exacly this line is responsible for my headaches
    }

I've tried to find out why it does not work, but my knowledge is not enough to solve the problem :(. Any help would be very appreciated :)

CornN64, 2023-01-05:
CrapFS::CrapFS app1 not found
Seems to indicate that it does not create the file system to store the ROMs in the ESP32 internal  EEPROM
I now this works for sure with Arduino 1.8.13 with board manager V1.0.4 and I had issues with newer board managers.
Did you try to run the NES or Atari console and did they work?

mikimause9, 2023-01-05:
Hi, I actually got it working! By trial and error I figured out that everything works after removing 'set_pref("recent",path);
' line. I have no idea what the problem is, but hey, it works :).

closed 2023-01-05 by mikimause9