rossumur/esp_8_bit #53: How to set proper CLOCK frequency?

panciunio, 2023-01-15:
Hi,

I compiled esp_8_bit with success and uploaded it to NodeMCU 32  module. I found pin with video signal, connected it to TV and no image appeared. Then I checked it on the oscilloscope and I see that time between each line is very short... 2uS.. as I remember, time to draw one line shall be about 64uS... I have no idea how to set proper value and where?
  
[image]

CornN64, 2023-01-21:
Indeed the time between sync pulses should be 64us, I am also not familiar with the NodeMCU 32 module...
If you have old silicon version of the ESP32 the APLL was broken, perhaps this is the reason(?).
In my version it outputs it in boot log if you have and older ESP32

panciunio, 2023-01-22:
Thanks for your replay. I checked it with a few modules e.g. LoLin 32, LoLin32 PRO MINI LiLyGO. Any tested has the same behavior. I'll try to find older revision... or use S2 module as well. I forgotten say THANKS! - It's very good and funny project, especially that Atari was my first computer :-)

CornN64, 2023-01-23:
make sure to compile it with board manager version 1.0.4. Newer ones have some issues

panciunio, 2023-01-24:
Yes. I'd used this rule... I tried to compile with newest library but I've got an error which was described in other topic. Let's go to test with other modules...