rossumur/esp_8_bit #60: Connecting Bluetooth controllers

Yan-007, 2023-09-29:
I read all contents of readme, but still can't understand how to connect a Bluetooth Wireless Controller.
There's no list of devices or something on the screen after booting, because ROM loads imediately.

RoCorbera, 2024-01-26:
What is the ESP32 Arduino Core version you are using to build it?

Try ESP32 Arduino Core 1.0.4

demik, 2024-06-28:
Is this a Wii controller ? There is no code for switch controllers

Yan-007, 2024-07-27:
> Is this a Wii controller ? There is no code for switch controllers

No, I was trying to connect Xbox controller (actually an alternative, but it shows up and working as Xbox does)

demik, 2024-08-08:
> No, I was trying to connect Xbox controller (actually an alternative, but it shows up and working as Xbox does)

To my limited knowledge that cannot work. The code only contains a driver for Wii controller