rossumur/esp_8_bit #65: SD-Card isn't working...

jojo-gfx, 2024-02-12:
Has anyone tried out the actual version with SD-Reader?
Have no frequency output on clock signal. It's standing only on 3,2V and so also ESP32 don't start really.
If I set SD-card off all other is running normal.

Have also tried out the last version here (without the config file) and it has the same problem.
If I connect a SD-Card reader, with or without SD-Card, the ESP32 won't start up.

... also I can't see any documentation about the SD-Card vonnection in this version!? Have had an older one and there was it documented but also not working.

Would be nice the project would be developed also in future, because I don't know such a nice thing with AV output. :(

jojo-gfx, 2024-02-12:
10:18:00.960 -> Rebooting...
10:18:00.960 -> ets Jul 29 2019 12:21:46
10:18:00.960 -> 
10:18:00.960 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
10:18:00.960 -> configsip: 0, SPIWP:0xee
10:18:00.960 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
10:18:00.960 -> mode:DIO, clock div:1
10:18:00.960 -> load:0x3fff0018,len:4

jojo-gfx, 2024-02-12:
Ok, this was without an SD-Card.
With SD-Card is the error so:
-> Rebooting...
10:36:55.348 -> E (376) sdmmc_sd: sdmmc_init_spi_crc: sdmmc_send_cmd_crc_on_off returned 0x106

jojo-gfx, 2024-02-12:
Ok, for all with such problems: Look on the write-protection! It has to be unlocked!
Now the serial output looks better, but still ni Output on AV and this: csum err:0xea!=0x07

jojo-gfx, 2024-02-12:
Ok, after format a same card per CMD format X: /FS:FAT /Q the following is coming up in console:

11:15:54.991 -> 
11:15:54.991 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
11:15:55.065 -> invalid header: 0xffffffff
11:15:55.110 -> invalid header: 0xffffffff
11:15:55.142 -> invalid header: 0xffffffff
11:15:55.188 -> invalid header: 0xffffffff
11:15:55.234 -> invalid header: 0xffffffff
11:15:55.266 -> invalid header: 0xffffffff
11:15:55.312 -> invalid header: 0xffffffff
11:15:55.357 -> ets Jul 29 2019 12:21:46
11:15:55.357 -> 

Seems, the card is read, but the format seems to be wrong.

I have to find now a Win XP or so to format it right. 
Seems the implementation of filesystem is really unstable. :(

franciscogimeno2000, 2024-09-20:
Hi everyone, first of all, thank you for this great project. One question, has anyone managed to emulate with an SD card? I've tried a lot of things and there's no way, the micro keeps rebooting. Regards.